microsoft/pxt-ev3 #839: Need RGB Raw values from color sensor

THEb0nny, 2019-03-13:
Please meake rgb raw mode. Nead RGB values.

THEb0nny, 2019-03-22:
```
_query() {
            if (this.mode == ColorSensorMode.RgbRaw)
                return this.getNumber(NumberFormat.UInt16LE, 0)
            return 0
        }

rgbraw() {
            this.setMode(ColorSensorMode.RgbRaw);
            let rgb: number[] = [];
            rgb[0] = this.getNumber(NumberFormat.UInt16LE, 0);
            rgb[1] = this.getNumber(NumberFormat.UInt16LE, 2);
            rgb[2] = this.getNumber(NumberFormat.UInt16LE, 4);
            return rgb;
        }
```

THEb0nny, 2019-04-05:
Pay attention to release this to release?

THEb0nny, 2019-06-06:
https://github.com/microsoft/pxt-ev3/pull/860

pelikhan, 2019-08-30:
in /beta

closed 2019-08-30 by pelikhan