microsoft/pxt-ev3 #840: Make it possible to write data to files

THEb0nny, 2019-03-15:
There is a need to store data and work with them through a text file in the memory of the controller.

pelikhan, 2019-03-15:
Have you tried adding the “storage” extension?

THEb0nny, 2019-03-16:
> Have you tried adding the “storage” extension?

Is there such a thing? Need a link to this project ...

THEb0nny, 2019-03-18:
> Have you tried adding the “storage” extension?

Какая ссылка на это расширение?
Kakaya ssylka na eto rasshireniye?
31/5000
What is the link to this extension?
Отправить отзыв
История
Сохранено
Сообщества

abchatra, 2019-04-12:
@pelikhan what storage are you referring to?

pelikhan, 2019-04-12:
storage package

abchatra, 2019-04-12:
I don't see any storage package appearing when I search in extensions.

pelikhan, 2019-04-12:
https://github.com/Microsoft/pxt-ev3/tree/master/libs/storage

maximmasterr, 2019-06-29:
Cannot connect repository using link above.
So I coppied files to my own repository and connect it using maximmasterr/pxt-ev3-storage
But I think that is not good way, so it is better to include it to dependency of ev3

THEb0nny, 2019-06-29:
> Cannot connect repository using link above.
> So I coppied files to my own repository and connect it using maximmasterr/pxt-ev3-storage
> But I think that is not good way, so it is better to include it to dependency of ev3

I could not add. Please go to discord makecode for a discussion.

pelikhan, 2019-08-28:
In /beta.

closed 2019-08-28 by pelikhan