microsoft/pxt-ev3 #841: The second sensor in the account is restarted at the start of the program

THEb0nny, 2019-03-17:
Good afternoon, I noticed that the color sensors start up very slowly. First, the account starts the first, and the second later. It generally restarts. And it can restart several times. It is in color reflection mode. If you set the color mode, the sensors are first included in the reflection mode, and then restarted to the color mode. It happens very slowly. We need to do something about it. And I propose to add a block / code that will wait until the sensor is working. Thank you.

THEb0nny, 2019-03-21:
```
_query() {
            if (this.mode == ColorSensorMode.RgbRaw)
                return this.getNumber(NumberFormat.UInt16LE, 0)
            return 0
        }

rgbraw() {
            this.setMode(ColorSensorMode.RgbRaw);
            let rgb: number[] = [];
            rgb[0] = this.getNumber(NumberFormat.UInt16LE, 0);
            rgb[1] = this.getNumber(NumberFormat.UInt16LE, 2);
            rgb[2] = this.getNumber(NumberFormat.UInt16LE, 4);
            return rgb;
        }
```

pelikhan, 2019-08-28:
Fix for https://github.com/microsoft/pxt-ev3/pull/887

pelikhan, 2019-08-28:
Fixed in /beta v1.1.9

closed 2019-08-28 by pelikhan