microsoft/pxt-ev3 #844: Tank Move turning right with rotational duration crashes program?

thebrucecgit, 2019-04-03:
# Bug Template

**Describe the bug**
When tank move is set with a power of "Left: 0" and "Right: 50" on any two motors with a rotational motor duration, it freezes the program and the motors are forever moving with Left: 0 and Right: 50. 

This works: 
`motors.largeAB.tank(50, 0, 1, MoveUnit.Rotations);`
This does not work: 
`motors.largeAB.tank(0, 50, 1, MoveUnit.Rotations);`

**To Reproduce**
Steps to reproduce the behavior:
1. Create a project on the Mindstorms Makecode website
2. Select the Javascript tab
3. Copy `motors.largeAB.tank(0, 50, 1, MoveUnit.Rotations);` into the project and run it. 
4. Nothing below it will run and the motors are stuck on that power. 

It works when the motor power is at Left: 50 and Right: 0 but not when it's Left: 0 and Right 50. It doesn't matter what number is in the rotation parameter but it only happens when it's `MoveUnit.Rotations`. 

**Expected behaviour**
The selected two motors are expected to run at the selected power of Left: 0 and Right: 50 for 1 rotation. (as it does when Left: 50, Right: 0)

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: MacOS Version 10.14.4 Beta
 - Browser: Chrome
 - Version: Version 74.0.3729.28 (Official Build) beta (64-bit)

**Additional context**
Add any other context about the problem here.

abchatra, 2019-04-03:
@thebrucecgit pxt-ev3 is the right repo for ev3 issues.

pelikhan, 2019-08-30:
Fix in https://makecode.mindstorms.com/beta v1.1.13 and above. YOu can use the tank tester example (https://makecode.mindstorms.com/reference/motors/synced/tank) to try out tank values.

closed 2019-08-30 by pelikhan