microsoft/pxt-ev3 #847: Multiple sensor issue

abchatra, 2019-04-12:
The robot acts funny when you have multiple sensors in at the same time. 

Bug - Feedback from FLL team which visited us.

THEb0nny, 2019-04-28:
Color sensors are still very long run ...

abchatra, 2019-04-29:
@THEb0nny I don't understand your comment. Can you elaborate?

pinwc4, 2019-04-29:
I was about to submit a bug and found this.  I suspect what I was going to submit is similar.

Testing with a singular sensor attached everything works great.  When you have multiple sensors it takes around 6 seconds before you can start reading data from them.  With just 1 sensor we can start a program and get data nearly immediately.  As soon as you add 1 or more additional sensors suddenly it takes 6 seconds before you start getting valid data back from them.  

It seems like there is some issue with initializing or detecting additional EV3 UART sensors (Color, Gyro, Ultrasonic).  I tested various combinations of those 3 sensors and it always occurs as soon as you have 2 or more of them.  

This does not happen if you only have 1 UART sensor in combination with a touch sensor, which is just an analog sensor.  

This will cause all sorts of unexpected behaviour in your program because it is getting 0's for sensor values and reacting to that instead of valid sensor data.  Then suddenly after 6 seconds it starts getting good data.  

Also your program does not have to actually try to read from the extra sensors to have this problem.  You could have a program that reads only from a color sensor but if you plug in a second one like a Gyro then suddenly it takes a long time to start reading data even if your program does not use the Gyro yet.  Just having them plugged into the EV3 is enough to have the issue.

abchatra, 2019-04-29:
@pelikhan fyi

THEb0nny, 2019-04-30:
> @THEb0nny I don't understand your comment. Can you elaborate?

Color sensors are initialized for a very long time when the program is started. I cannot start the necessary mode for some time, but I’m generally afraid to imagine what will happen about switching modes at the moment of work. Sometimes it even happens that the sensor can turn off.

THEb0nny, 2019-04-30:
pinwc4 principle painted a problem. I don’t know 6 seconds, but the color sensors first turn on the reflection mode (red), blink, if everything is fine, they stop flashing, but they can continue to blink for a long time ...

maximmasterr, 2019-06-28:
I think this problem is related to set all sensors to None(-1) mode and sensors cant exit from this mode
Maybe setting None to 0 will fix problem

maximmasterr, 2019-06-28:
You can wait untill sensors will be ready
```ts
pauseUntil(() => sensors.color2.light(LightIntensityMode.Ambient) != 0 && sensors.color3.light(LightIntensityMode.Ambient) != 0)
```

maximmasterr, 2019-06-28:
Just add those lines to start of your code
```ts
sensors.color2.reset()
sensors.color3.reset()
```
One sensor will turn off and ater second turn on
Then they wil work correctly

THEb0nny, 2019-06-28:
> Just add those lines to start of your code
> 
> ```ts
> sensors.color2.reset()
> sensors.color3.reset()
> ```
> 
> One sensor will turn off and ater second turn on
> Then they wil work correctly
> I think the problem is in libs/core/input.ts:215
> It need to be reset to work correctly
I still used the option.

```
sensors.color2.color(); sensors.color3.color();
pauseUntil(() => sensors.color2.isActive() && sensors.color3.isActive());
brick.showString("   RUN", 6);
while (!brick.buttonEnter.wasPressed()) pause(10);
```

maximmasterr, 2019-07-03:
#868

maximmasterr, 2019-08-04:
The problem is in reseting sensors

abchatra, 2019-08-29:
@pinwc4 Share your program please.

pinwc4, 2019-08-29:
@abchatra Sure, I just made a simple program to test this easily.  

`forever(function () {
    brick.showValue("Color1", sensors.color1.light(LightIntensityMode.Reflected), 1)
    brick.showValue("Color2", sensors.color2.light(LightIntensityMode.Reflected), 2)
    brick.showValue("Sonic1", sensors.ultrasonic3.distance(), 3)
    brick.showValue("Sonic2", sensors.ultrasonic4.distance(), 4)
})`

pinwc4, 2019-08-29:
I saw some commits that were around sensor inputs so I tested this again on beta 1.1.11.  It is working correctly now for me, all sensors starting reading data immediately on program start.  Thank you for all your work on this