microsoft/pxt-ev3 #849: Exit out of program

abchatra, 2019-04-12:
Sometimes we can't exit out of the program (Abhijith: Feedback likely on the brick)
Bug - Feedback from FLL team which visited us.

abchatra, 2019-04-12:
May be they had infinite while loop or something

THEb0nny, 2019-04-28:
Such a problem exists and it is necessary to solve it yes.

abchatra, 2019-04-29:
@THEb0nny Can you share the program which you are unable to exit?

THEb0nny, 2019-04-30:
Now there is no code for such a program, but I have had this many times. Occurs in a loop. I had to make a condition to press any button (but there is no access to the button back: ((())

maximmasterr, 2019-08-28:
Look here: https://github.com/microsoft/pxt-ev3/pull/879#issuecomment-520343421

pelikhan, 2019-08-28:
@mmoskal it seems that if the readButton loop does not get run with a tight loop.

mmoskal, 2019-08-29:
I think it may be that the loop in runPoller() is stuck at startUser() after previous button update, and so the readButtons doesn't run.

we probably need to have a version of startUser with a timeout using https://linux.die.net/man/3/pthread_mutex_timedlock and use that in the runPoller(). Then if it timeouts, we should run query again I guess.

mmoskal, 2019-08-29:
or maybe better have a completely separate thread just polling the exit button, like usbThread or pumpMusicThread

maximmasterr, 2019-08-29:
There is problem 162-167 need to be runned in parallel
https://github.com/microsoft/pxt-ev3/blob/fd75bb61d6100b9d7dfb11f80d5683ea2ee398c1/libs/core/buttons.ts#L162-L167

pelikhan, 2019-08-29:
Fix in /beta v1.1.10

PR #888 (2019-08-29): Add exitThread() polling the exit button
Not tested.

Should fix #849
closed 2019-08-29 by pelikhan