microsoft/pxt-ev3 #852: Link not working

tiestapistics, 2019-04-22:
When you create a custom block and publish it via GitHub, a README file is created. The links to e.g. send a package to the Minestorm approval do not work:

https://makecode.mindstorms.com/packages/approval
https://makecode.mindstorms.com/packages/build-your-own

Our project is: https://github.com/tiestapistics/EV3_drive

abchatra, 2019-04-22:
@tiestapistics Can you elaborate your issue? Are you trying to build an extension? 

Loading the extension (https://github.com/tiestapistics/EV3_drive)  through the extensions UI seems to be working fine.  You need to copy past the github url to the extensions search bar.

tiestapistics, 2019-04-22:
Yes, you're right. The extension itself works perfectly.
But the problem I reported is one step further. When the extension was uploaded, a README.md was automatically created with some TODO's that you should customize to complete this extension. Some of the links that were automatically created in the README.md are unfortunately not available.

Here's what I mean:

TODO
* Add a reference for your blocks here
* Add "icon.png" image (300x200) in the root folder
* Add "- beta" to the GitHub project description if you are still iterating it.
* Turn on your automated build on https://travis-ci.org
* Use "pxt bump" to create a tagged release on GitHub
* Get your package reviewed and approved https://makecode.mindstorms.com/packages/approval
Read more at https://makecode.mindstorms.com/packages/build-your-own

pelikhan, 2019-08-30:
Thank you, the links have been fixed

closed 2019-08-30 by pelikhan