microsoft/pxt-ev3 #854: Tankmotor reset when using two instances

pinwc4, 2019-04-23:
We attempted to use the tankmotor block for 2 sets of motors at the same time but it fails.  When the second tankmotor block gets called it stops the action of the first.  So for example if the first tankmotor block sets motors AD to 50% and then the second tankmotor block sets motors BC to 75% for 3 rotations motors AD will stop.  They should continue running at 50% but it appears as thought he second tankmotor call overrides the first one.  

The simulator shows the correct behaviour but when you upload your code the EV3 motors do not respond the way the simulator shows.

Steps to reproduce the behavior:
1. Add 3 blocks to your Forever loop.  Block 1 can be a tankmotor AD set to -70% forever.  Block 2 will be tankmotor BC to 50% for 3 rotations and Block 3 will be tankmotor AD to 50% for 3 rotations.

Expected behavior:
The simulator will show this program correctly.  Motors AD will run at -70% while motors BC run at 50% for 3 rotations.  Then when the rotations complete motors AD will run at 50% for 3 rotations.  After that loop completes it starts over setting AD to -70% while BC run for 3 rotations at 50% etc.  

Actual behavior:
The uploaded code to the EV3 does not behave like the simulation.  Motors AD never run at -70% or at least not for long enough to ever be noticed.  

Example code:
forever(function () {
    motors.largeAD.tank(-75, -75)
    motors.largeBC.tank(50, 50, 3, MoveUnit.Rotations)
    motors.largeAD.tank(50, 50, 3, MoveUnit.Rotations)
})

Environment:
OS: Windows 10
Browser: Chrome
Controller: EV3

pelikhan, 2019-08-29:
After testing with labview, it seems that only 1 tank block is supported at any point.

pelikhan, 2019-08-30:
The simulator behavior has been updated to reflect the hardware behavior. Only one sync speed command can run at any time. Thank you for reporting.

pelikhan, 2019-08-30:
You can test this at https://makecode.mindstorms.com/beta

closed 2019-08-30 by pelikhan