microsoft/pxt-ev3 #857: try / catch statement

tiestapistics, 2019-05-01:
We are missing a feature like a try / catch statement like in JavaScript:
For example:

try {
  // your code here
  // pause(1*1000);

  throw("Something goes wrong");

  // code not executed
} catch (Exception e) {
  let str: = e->getMessage();
}

All pause() and control.waitMicros() functions should be stopped immediately. It should be possible to stop motors or move motors to a safe position in the catch block.

Other ideas:

The panic() function is already a very interesting approach, but it terminates the whole program, and you can't jump back to the main loop of your program. Another problem with panic() is that the moving motors are not stopped.

Unfortunately, the code executed within control.runInBackground() / control.runInParallel() cannot be aborted e.g. by the touch sensor event.

tiestapistics, 2019-05-05:
[image]

The LEGO MINDSTORMS EV3-Software provides the ability to end a loop and jump to a named parent named loop. This concept is very similar to the try / catch concept known from JavaScript. Whereby you could also compare it with a goto construct often ridiculed by programmers.

However, it would be really desirable to have a construct in MakeCode, to be able to jump from functions and loops in the event of an error, and to put the system back into a good state with superior error handling.

THEb0nny, 2019-06-22:
try catch need

pelikhan, 2019-10-10:
@tiestapistics we provide a "break" and "continue" block that allows to break out of a loop or continue to the next iteration.

pelikhan, 2019-10-10:
@THEb0nny try/catch in /beta

closed 2019-10-10 by pelikhan