microsoft/pxt-ev3 #858: MacOS download of offline app doesn't unzip

jimbobbennett, 2019-05-02:
**Describe the bug**
The zip file for the offline editor for Mac doesn't unzip, it gives an error.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://makecode.microbit.org/offline-app
2. Agree the terms and conditions
3. Download the MacOS app
4. It will fail to unzip

**Expected behavior**
The download should unzip successfully

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: MacOS 10.14.4 (Mojave)
 - Browser: Safari
 - Version: 12.1

**Additional context**
Add any other context about the problem here.

abchatra, 2019-05-02:
@jwunderl can you try to reproduce this?

jwunderl, 2019-05-02:
It works for me when downloading it from chrome and manually unzipping it, but failed the first time I tried downloading from safari (though trying to download it again consistently produced valid apps). 

For completeness; mine's a MBP 15in 2018 / 10.14.4 (Mojave), using `unarchiver` as my default unzip utility.

abchatra, 2019-05-02:
@jimbobbennett does it fail to download repeatedly?

jimbobbennett, 2019-05-03:
Yup - fails every time on Safari.

[image]

Fails on Chrome as well - the downloads can't be unzipped by the default MacOS unzip tool - Archive Utility..

pelikhan, 2019-08-30:
@abchatra is this fixed?

abchatra, 2019-08-30:
This got fixed in microbit when we released a new payload. Likely cdn caching issue.

pelikhan, 2019-10-10:
@jimbobbennett please try again and reopen if the issue persists.

closed 2019-10-10 by pelikhan