microsoft/pxt-ev3 #859: MakeCode version and EV3 firmware version

tiestapistics, 2019-05-02:
There should be a MakeCode version number that can be read in the source code. E.g. to find out if a certain bug with the motor control in MakeCode has already been fixed. One would like to see in the source code, for example, whether the beta version (https://makecode.mindstorms.com/beta) of MakeCode is used.

Also, the firmware version of the EV3 would be interesting for the future to know.

With control.deviceSerialNumber() you can see which brick is connected, but you have no information about the firmware and the MakeCode version.

abchatra, 2019-06-03:
We strive for backward compat. I am not sure how the version will help. We also combine entire runtime with the .uf2 file. So version of the MakeCode is always tied to the runtime. Please give us scenarios to understand.

tiestapistics, 2019-06-04:
It is good that the runtime is contained in a .uf2 file. But especially when developing a library, you want a certain bug in the runtime environment to be fixed. The user can install the runtime environment as Windows-APP, there is the BETA version and finally also the official web version. After it has been compiled, you can no longer tell where and how the source code has been compiled. Maybe a compilation date would be enough instead of a version number. Especially for the First Lego League competition, it would help if you could see which is the last compiled version.