microsoft/pxt-ev3 #862: Bluetooth

maximmasterr, 2019-05-28:
Bluetooth support beetwen robots

abchatra, 2019-06-03:
Please give a clear description of the feature request. Use feature request template for filing issues.

maximmasterr, 2019-06-12:
This feature related to create robots that communicates beetwen each one over bluetooth

### Solution
Built-in feature or extension to support communication over bluetooth beetwen two or more ev3
```js
//Example for sending string to ev3 with name Ev3-1
bluetooth.send('Ev3-1','Example string')

//Example of recieving data
bluetooth.on('data',(a=data)=>console.log(data))
```

### Alternatives
Other way to comunicate beetwen bricks

maximmasterr, 2019-07-13:
Need to update compilation method add -lbluetooth

maximmasterr, 2019-08-29:
Are you going to release this

pelikhan, 2019-08-29:
Unlikely that we implement this feature this year but we do take pull requests.

pelikhan, 2019-10-02:
See https://makecode.mindstorms.com/bluetooth

closed 2019-10-02 by pelikhan

maximmasterr, 2019-10-03:
Not I mean communication beetwen robots, not between robot and computer, reopen ut

maximmasterr, 2019-10-03:
Reopen

Beat-YT, 2021-07-23:
Yea