microsoft/pxt-ev3 #865: Gyro doesn't work as intended

abchatra, 2019-06-06:
Hi, I am a robotics class teacher. We are using the Lego EV3 system and we are programming it in MakeCode.
I have troubles with the Gyro sensor. Whenever I try to read out the angle value the whole system is slow/not working. Even the example code in the official reference does not work. See:
https://makecode.mindstorms.com/reference/sensors/gyro/angle
If you run this example, you will find out that it does not work.
This is very frustrating and disapointing. It is not easy tu explain to my students that the example code in official reference does not work... Do you have any idea how to solve this? Any solution or workaround?
Thank you and have a nice day.

pelikhan, 2019-08-30:
Fix in /beta v1.1.14.

closed 2019-08-30 by pelikhan

mjb-cal, 2019-09-16:
Sorry if I'm not following protocol but I'm seeing v1.1.20 and the EV3 gyro issues as described above are not resolved. 

Fully tested hardware using Lego Mindstorms software and it all works as it should.  In Makecode the Gyro reads 0 continually, occasionally-1.

Any help appreciated.

abchatra, 2019-09-16:
@mjb-cal can you share your program? Use share button at the top of the editor.