microsoft/pxt-ev3 #869: Command-V Does Note Paste Code in Offline App Macintosh Version only

MVSLSFLL, 2019-07-06:
**Describe the bug**
There is no ability to paste text or code that has been copied to the clipboard using Command-V in the Macintosh version of the MakeCode Mindstorms offline app. The browser version responds to Command-V and the Windows version of the offline app responds to Cntl-V 

**To Reproduce**
Steps to reproduce the behavior:
1. Open a project
2. Toggle the JavaScript switch
3. Copy any text/code in the editor window using Command-C or the Copy function through the mouse.
4. Press Command-V anywhere and the text/code is not pasted in to the editor

**Expected behavior**
As in the Windows offline app and the browser versions of the app, I would expect to see the text pasted into the editor wherever my cursor was placed.

**Screenshots**
Add screenshots to help explain your problem. You can copy paste the screenshot in the github report. The .gif screen recording is very useful as well. 

**Desktop (please complete the following information):**
 - OS: MacOS Mojave 10.14.5
 - Browser: MacOS version of offline app
 - Version: Mindstorms EV3 version 1.0.11
                  MakeCode version 4.0.9

**Context**
- It would be extremely useful to get this fixed as the next season of FIRST Lego League is starting on Aug 1, 2019.

maximmasterr, 2019-07-09:
I think it`s better to use web application, command-v work correctly there and it`s offline to (It will work without interne)

MVSLSFLL, 2019-07-09:
If I understand the message right, I can use the web app without actually being connected to the internet? I'll try that. The problem seems to only be on MacOS and most of the FLL team uses Windows.

Thanks for the response.

maximmasterr, 2019-07-14:
Yes you understand it right, you can use it without connection to internet. I also recomend you to use chrome instead safari, and creating bookmark to navigate to [https://makecode.mindstorms.com/](https://makecode.mindstorms.com/)

Technonati, 2019-07-14:
Great. Thank you for the help.

Mike

closed 2019-08-21 by Technonati