microsoft/pxt-ev3 #873: Release plan

tiestapistics, 2019-07-19:
Hi,

is there already a plan for a next release? At the beginning of August the FIRST LEGO League competition will start in Europe. So it would be important to know when we can expect certain things like a try/catch block to be implemented by then.

The following information is currently available on the https://makecode.mindstorms.com/ website: 
  LEGO® MINDSTORMS® Education EV3 version: 1.0.11
  Microsoft MakeCode version: 4.0.9

MakeCode version 4.0.9 is from September 26, 2018 and there is already version 5.18.9 available. And pxt-ev3 version 1.0.11 is now already half a year old without updates.  And even on the site https://makecode.mindstorms.com/beta there are no new versions.

For FLL it would be desirable if there were regular releases and a release plan. At least some information about the future of the project would be desirable.

abchatra, 2019-08-06:
We release once a year release on many editors. For example: microbit. This is mostly to keep the editor stable for teachers.  We strive for backward compat as well. You can take dependency on the current editor and need not wait for the new editor. All features will continue to work on new release as well. 

Try\catch is brand new feature and will like have lot of bugs. We will wait for it to stabilize before shipping.

Next release for ev3 will be by end of this year. There is no fixed timeframe yet.

closed 2019-08-06 by abchatra

Technonati, 2019-08-22:
MakeCode Mindstorms is pretty much unusable for FLL basic file management capabilities on the brick need to be addressed, as well as how the program you download starts running immediately when you drag it to the brick.

abchatra, 2019-08-22:
Running immediately is by design (this gives instant gratification to students). All our editors do that. Can you explain why is it undesirable for your scenario?

abchatra, 2019-08-22:
There are always events one could use (like on button pressed) to control the invocation of your code.

pelikhan, 2019-08-30:
@Technonati we have pushed a number of fixes on https://makecode.mindstorms.com/beta . Please try them out.
* "storage" extension to write files
* better gyro calibration
* fixes to tank blocks
* fixes to handling of escape button, etc..

Technonati, 2019-08-30:
Thank you. We will try it out.

I realize it a monumental task to build an programming environment, and my
comments were all given with the greatest respect for the work you are all
doing.

MakeCode could be a really cool tool for the thousands of FLL teams that
compete each year in FLL. It is a lot cleaner to program with the MakeCode
blocks than EV3-G (LabView).

pelikhan, 2019-08-30:
We are actively pushing fixes to /beta right now — also we are looking for additional examples and tutorials rom FLL coaches ;)

Technonati, 2019-09-01:
Understood...and thank you, again.