microsoft/pxt-ev3 #881: Missing main.blocks file when creating a repo using Import

Technonati, 2019-08-15:
**Describe the bug**
I created an Extension using GitHub using the instructions at https://makecode.com/extensions/github-authoring.

I now want to use the extension in a EV3 program, but I also want to have my MakeCode Mindstorms programs saved on Github as repos. 

But when I generate a repo for the program from using Import/My Github Repos, no main.blocks file is created and I can't use the Blocks view to program. The generated repo open in Javascript view (as it should) but there is no blocks view

**To Reproduce**
Go to https://makecode.mindstorms.com/
(Assuming your Github account is connected already)
Click Home
Click on the Import button
Click on the Your Github Repo... box
Click Create New... option
Name your repo
Click on the Go ahead! button
MakeCode editor opens
Blocks/JavaScript slider at top of window slides to JavaScript
Editor shows blank contents of main.ts file with the Explorer sidebar open and the main.ts file selected.
Attempting to move the Blocks/JavaScript slider is unsuccessful.
Looking at explorer, no main.blocks file is present.

**Expected behavior**
I would have expected a main.blocks file to be generated that would allow you to switch from Javascript view to Blocks view to allow programming with blocks.

**Screenshots**
Add screenshots to help explain your problem. You can copy paste the screenshot in the github report. The .gif screen recording is very useful as well. 

**Desktop (please complete the following information):**
 - OS: Mac OS
 - Browser: Chrome
 - Version 76.0.3809.100

**Additional context**
I understand there is no need for a Blocks view inside of an extension. It would also be nice to be able to use Github to store the EV3 programs we make that use our extensions, so the 10 kids I have programming on the task can maintain version control. This all may be operator error and I have just missed reading the right instructions somewhere online.

If I have to fatfinger the main.blocks files in the program repo I create, I can accept that, but all attempts at that have resulted in a "Network Request Failed" message when loading the fatfingered repo.

cronello, 2019-09-04:
Hi, did you solve it somehow? I'd like to use git as versioning tool for common makecode projects (for micro:bit) for kids also; and go through the same way - create repo for extension but missing the blocks "programming".
I'm appreciated for any help if it is possible. Thx

Technonati, 2019-09-05:
These are the steps I recall:

1. In MakeCode, after linking my Github account, I back out and use Import
to create a repo, so all the key files are there.
2. In MakeCode, I go into Javascript view and hit the + and create a
custom.ts file, and then sync, and exit MakeCode
3. In Github, I change the custom.ts file name to main.blocks and in
pxt.json I change the reference to custom.ts to main.blocks
4. In Github, I copypasta the following XML script into main.blocks:

<xml xmlns="http://www.w3.org/1999/xhtml">
<variables></variables>
<block type="forever" x="0" y="0"></block>
<block type="pxt-on-start" x="180" y="0"></block>
</xml>

Try it out. If you have any questions, let me know and I'll go back through
my notes.

Cheers.