microsoft/pxt-ev3 #898: Motor schedule

maximmasterr, 2019-08-31:
Sometimes just run motor for 100 degrees is not enough because of inertia, so we need to create motor schedule it is 3 staged motor control 
1. Acceleration stage 
2. Normal stage
3. Deaceleration stage

```ts
schedule(speed: number, acceleration: number, value: number, deacceleration: number, unit: MoveUnit)
```

Like run it has speed, value, and unit  

Also there are two other parameters  
acceleration distance in `units` for acceleration and
deacceleration distance in `units` for deacceleratio

This feature is already implemented in ev3

This is acceleration length: 
https://github.com/microsoft/pxt-ev3/blob/d5194b8d28e163b4c5cc90068b60d23de0dbb636/libs/core/output.ts#L719

This is deacceleration length: 
https://github.com/microsoft/pxt-ev3/blob/d5194b8d28e163b4c5cc90068b60d23de0dbb636/libs/core/output.ts#L721

It is recomended to realise this feature both for single and two motors

maximmasterr, 2019-08-31:
Example of shchedule can be found in ev3 basic https://sites.google.com/site/ev3basic/ev3-basic-programming/using-motors

pelikhan, 2019-09-03:
in beta.

closed 2019-09-03 by pelikhan

mrtbart, 2020-01-18:
While experimenting I found some strange behavior of Motor.Schedule
The next example is meant to make a square with the base-model Lego robot
The first of the four forward movements is OK, but in the 2, 3, and 4th movement the wheels are not driven simultaneously, but one after another (B first, than C)

For i=1 to 4
  Motor.Schedule("BC", 60, 180, 360, 180, "True")  
  Motor.Wait ("BC")
  Motor.MoveSync("BC", 40, -40, 184, "True")
EndFor