microsoft/pxt-ev3 #939: Working with NXT sensors

RasPiBot, 2019-10-08:
Hello MakeCode community,

I'm a computer science teacher in Quebec Canada and we use MakeCode in our introduction to computer programming course. We have 24 **EV3** bricks with **NXT** motors and sensors.

Is it possible that MakeCode doesn't work will all nxt sensors? My code is running well in the simulator then nothing happen on the real robot. I tried with touch and ultrasonic sensors.

We are new to MakeCode, we were using MonoBrick, Mindsqualls and Lejos before that.

Thanks,

Shany

abchatra, 2019-10-09:
We only support EV3 and not NXT. NXT doesn't have firmware support for MakeCode.

abchatra, 2019-10-09:
@pelikhan are NXT sensors compatible with EV3 brick? If yes NXT sensors should work as well with MakeCode.

THEb0nny, 2019-10-09:
> @pelikhan are NXT sensors compatible with EV3 brick? If yes NXT sensors should work as well with MakeCode.

Sensors compatible with ev3. We should have access to the nxt light sensor and the temperature sensor, maybe I'll deal with it.

THEb0nny, 2024-02-04:
I have implemented support for the nxt light sensor. I also have a temperature sensor, but no other sensors.