microsoft/pxt-ev3 #963: Missing large fonts for EV3 display

OliviliK, 2020-01-01:
**Describe the bug**
With LabVIEW, the program can show text on 12 lines of small fonts or 6 lines of double size fonts. PXT-EV3 is missing this feature.

**To Reproduce**
Steps to reproduce the behavior:
Open the Brick menu in block editor and observe that the display font size selection is not available.

**Expected behavior**
Similar functionality that was available with LabVIEW.

**Additional context**
In addition of having 12 or 6 lines of text, it is desirable to have 4 or 3 lines of text on display.  These lines would have fonts with height of 1/12, 2/12, 3/12, or 4/12 of the display height.

OliviliK, 2020-03-18:
Because my teams do need this feature, I created an a temporary solution for large numbers,  It did require 12 lines for the fonts and 13 lines to draw the pixels.  It is working fine, but is not an ideal solution.

pelikhan, 2020-03-19:
Could you share it in a GitHub repository?

OliviliK, 2020-03-19:
The custom.ts file can be found at [here](https://github.com/OliviliK/MakeCode/tree/master/EV3/WVPL).  Note the lines 454 - 466 to define a drawing canvas and 10x8 fonts for the 10 digits.

`let _canvas = image.create(screen.width, screen.height) // 178 x 128`
`let _digitFont = [  // pixels in 10 rows and 8 columns for each digit`
`    /* 0 */ 0x38, 0x6C, 0xC6, 0xC6, 0xC6, 0xC6, 0xC6, 0xC6, 0x6C, 0x38,`
`    /* 1 */ 0x18, 0x38, 0x78, 0xD8, 0x18, 0x18, 0x18, 0x18, 0x18, 0xFF,`
`    /* 2 */ 0x38, 0x6C, 0xC6, 0x06, 0x06, 0x0C, 0x18, 0x30, 0x60, 0xFE,`
`    /* 3 */ 0xFE, 0x06, 0x0C, 0x18, 0x38, 0x0C, 0x06, 0x06, 0xCC, 0x78,`
`    /* 4 */ 0x0C, 0x1C, 0x3C, 0x6C, 0xCC, 0xCC, 0xFE, 0x0C, 0x0C, 0x0C,`
`    /* 5 */ 0xFE, 0xC0, 0xC0, 0xC0, 0xF8, 0x0C, 0x06, 0x06, 0xCC, 0x78,`
`    /* 6 */ 0x3C, 0x60, 0xC0, 0xC0, 0xF8, 0xCC, 0xC6, 0xC6, 0x6C, 0x38,`
`    /* 7 */ 0xFE, 0x06, 0x0C, 0x0C, 0x18, 0x18, 0x30, 0x30, 0x30, 0x30,`
`    /* 8 */ 0x38, 0x6C, 0xC6, 0x6C, 0x38, 0x6C, 0xC6, 0xC6, 0x6C, 0x38,`
`    /* 9 */ 0x38, 0x6C, 0xC6, 0xC6, 0x6E, 0x3E, 0x06, 0x06, 0x0C, 0x78`
`]`

The function to draw a digit is on lines 525 - 537.

`function drawDigit(x: number, y: number, digit: number) {`
`    for (let i = 0; i < 10; i++) {      // For each row in the digit`
`        let byte = _digitFont[digit * 10 + i]`
`        for (let j = 0; j < 8; j++) {   // For each column in the digit`
`            if (byte >= 128) {          // Detect the next visible pixel`
`                byte -= 128`
`                                        // Draw a 3x3 box in 2x2 grid`
`                _canvas.fillRect(x + 2 * j, y + 2 * i, 3, 3, 1)`
`            }`
`            byte *= 2`
`        }`
`    }`
`}`

Sorry, that the formatting is not proper in this comment.

**Note:** This function is part of the 

`export function showMenu(index: number, title: string, labels: string) {`

function on line 782 used to show navigation menu.  This function labels all the 5 available buttons in the EV3 brick.