microsoft/pxt-ev3 #976: Storage Extension

qqa12, 2020-02-29:
The (beta) storage extension doesn’t work at all. It isn’t able to write to a file or read from it. Neither with USB drives nor with the internal brick storage.

THEb0nny, 2021-03-21:
I already wrote about it.
I cannot write to a file and read on the stable version.

```
const fileName = "colors_sensors_values.txt";
let whiteLeftColorS = 0, blackLeftColorS = 0;
let whiteRightColorS = 0, blackRightColorS = 0;

function Main() {
    for (let i = 0; i < 4; i++) {
        let k = 0;
        let sensor = (i <= 2 ? sensors.color2 : sensors.color3);
        let colorSensorSum = 0;
        while (true) {
            brick.clearScreen();
            brick.showString("Calibrate " + (i <= 1 ? "LEFT" : "RIGHT") + " ColorS", 1);
            let sensorVal = sensor.light(LightIntensityMode.ReflectedRaw);
            brick.showValue("Sensor val", sensorVal, 3);
            brick.showString("Press \"DOWN\" btn ", 6);
            brick.showString("to save " + (i % 2 == 0 ? "WHITE" : "BLACK") + " value", 7);
            brick.showString("\"ENTER\" to next.", 8);
            if (brick.buttonDown.wasPressed()) {
                colorSensorSum += sensorVal;
                k++;
                control.runInParallel(function () {
                    music.playTone(262, music.beat(BeatFraction.Half));
                });
                loops.pause(500);
            } else if (k >= 2 && brick.buttonEnter.wasPressed()) break;
            loops.pause(10);
        }
        let tmpValue = Math.round(colorSensorSum / k);
        if (i == 0) whiteLeftColorS = tmpValue;
        else if (i == 1) blackLeftColorS = tmpValue;
        if (i == 2) whiteRightColorS = tmpValue;
        else if (i == 3) blackRightColorS = tmpValue;
        loops.pause(200);
    }
    if (storage.permanent.exists(fileName)) storage.permanent.remove(fileName);
    storage.permanent.append(fileName, "");
    storage.permanent.appendLine(fileName, whiteLeftColorS + " " + blackLeftColorS);
    storage.permanent.appendLine(fileName, whiteRightColorS + " " + blackRightColorS);
    brick.clearScreen();
    brick.showString("Finish!", 1);
}

Main(); // Запуск главной функции
```

```
const fileName = "colors_sensors_values.txt";
let whiteLeftColorS = 0, blackLeftColorS = 0;
let whiteRightColorS = 0, blackRightColorS = 0;

function Main() {
    brick.clearScreen();
    if (storage.permanent.exists(fileName)) {
        brick.showString("File exists", 1);
    } else {
        brick.showString("File not!", 1);
    }
    loops.pause(10000);
}

Main(); // Запуск главной функции```

Writes that there is no file.

THEb0nny, 2023-08-28:
In my pxt v9 branch, the storage extension issue is fixed.