microsoft/pxt-ev3 #980: Cannot reliably mount emulated storage under Linux

JakubVanek, 2020-03-15:
**Describe the bug**
Linux-based OSes cannot reliably mount the emulated FAT32 volume. The underlying block device *is* accessible, but the FAT32 format cannot be parsed without errors.

**To Reproduce**
1. Boot Xubuntu 18.04.4 Live USB
2. Plug in EV3 with firmware 1.10E
3. Open the device in Thunar/File browser
4. Run `dmesg` in terminal

**Expected behavior**
Kernel mounts the filesystem successfully and is able to read and write files on it without errors.

**Screenshots**
dmesg on PC:
```
[46037.916480] usb 1-9: new high-speed USB device number 59 using xhci_hcd
[46038.104311] usb 1-9: New USB device found, idVendor=0694, idProduct=0005, bcdDevice= 2.16
[46038.104315] usb 1-9: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[46038.104317] usb 1-9: Product: EV3
[46038.104319] usb 1-9: Manufacturer: LEGO Group
[46038.104320] usb 1-9: SerialNumber: 00165344cf21
[46038.149724] hid-generic 0003:0694:0005.002E: hiddev1,hidraw0: USB HID v1.10 Device [LEGO Group EV3] on usb-0000:00:14.0-9/input0
[46038.157223] usb-storage 1-9:1.1: USB Mass Storage device detected
[46038.157599] scsi host6: usb-storage 1-9:1.1
[46049.321161] scsi 6:0:0:0: Direct-Access     Linux    File-CD Gadget   0316 PQ: 0 ANSI: 2
[46049.321848] sd 6:0:0:0: Attached scsi generic sg2 type 0
[46049.322918] sd 6:0:0:0: Power-on or device reset occurred
[46049.323847] sd 6:0:0:0: [sdb] 65000 512-byte logical blocks: (33.3 MB/31.7 MiB)
[46049.324376] sd 6:0:0:0: [sdb] Write Protect is off
[46049.324383] sd 6:0:0:0: [sdb] Mode Sense: 0f 00 00 00
[46049.324989] sd 6:0:0:0: [sdb] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[46049.350182]  sdb:
[46049.353691] sd 6:0:0:0: [sdb] Attached SCSI removable disk
[46056.286985] FAT-fs (sdb): error, corrupted directory (invalid entries)
[46056.286989] FAT-fs (sdb): Filesystem has been set read-only
[46088.069243] FAT-fs (sdb): error, fat_get_cluster: invalid cluster chain (i_pos 0)
```

If needed, I can scrape the kernel dmesg from the brick as well (via telnet).

**Desktop (please complete the following information):**
 - OS: Xubuntu 18.04.4
 - Version `5.3.0-40-generic #32~18.04.1-Ubuntu`

**Additional context**
I can force the kernel into ignoring the errors. After that, the files can be listed and application upload succeeds, but it is not a very good experience.
```
$ sudo mount -t vfat -o rw,errors=continue /dev/sdb /mnt
$ LC_ALL=C ls
ls: cannot access 'projects': Input/output error
dmesg.txt  index.htm  info_uf2.txt  projects
$ sudo cp lego-Untitled.uf2 /mnt/
$ sync
```

However I cannot access the `projects` directory at all, the result is always an I/O error. On Windows, I can browser the Projects directory and the UF2 files show up.

JakubVanek, 2020-03-15:
Snapshot of the raw contents of the emulated USB drive: [ev3fs.zip](https://github.com/microsoft/pxt-ev3/files/4335365/ev3fs.zip)

JakubVanek, 2020-03-16:
This seems to be a problem with the `Projects` folder only. If I remove all files from the `BrkProg_SAVE` directory, the folder disappears and there are no more errors while mounting.

JakubVanek, 2020-03-16:
The issue can be reproduced with https://github.com/microsoft/uf2-linux/, so it is not specific to EV3.
With the following directory map, the same errors occur via NBD.
```c
struct DirMap dirMaps[] = { //
    {"Uf2Dir", "/home/kuba/uf2", F_UF2},
    {NULL, NULL, 0}};
```

luukp, 2021-06-29:
I am also having this issue. Possibly the documentation should be updated to state that drag and drop isn't supported on Linux? I'm assuming that this will affect chromebooks as well, which kind of seems like the primary use case.

As this is likely to be a bug with the EV3 firmware, rather than  a problem with makecode itself, should it be raised with Lego to be resolved?

erwroth, 2021-07-17:
Same problem here on a Raspberry Pi 400 with Raspbian GNU/Linux 10.Would be great for teaching if this problem could be fixed soon!

tentator, 2021-12-28:
same here :(

lots of kernel: FAT-fs errors, first this:
    error, corrupted directory (invalid entries)  
    Filesystem has been set read-only
and then lots of these at each access:
    error, fat_get_cluster: invalid cluster chain (i_pos 0)

but it seems I cannot see the drive in windows as well and do not see the ev3 in the firmware upgrade tool anymore (otherwise I'd try to downgrade it for instance)..

I also found this alternative explanation/root cause so far: https://serverfault.com/questions/331779/how-do-i-debug-this-fs-error-on-a-flash-device

anyone has some progress on this?