microsoft/pxt-ev3 #981: There is no way to switch the type of engines

THEb0nny, 2020-03-17:
**Describe the bug**
There is no way to switch the type of engines. This is manifested in languages other than English. On stable and beta.

**To Reproduce**
Video https://drive.google.com/open?id=1ZUOegPX5_n17kL_yu1nxge1N41kjg435

**Expected behavior**
Must open a list.

MKleinSB, 2020-03-23:
@THEb0nny Did you translate large motor / medium motor / large motors ?

THEb0nny, 2020-03-23:
@MKleinSB I originally translated. Later deleted the translation. When I recorded a video with a bug, there was no translation. Then he added again.

MKleinSB, 2020-03-24:
@pelikhan
once you translated  large motor / medium motor / large motors you cannot change the motortype any more. 
The Blocks hangs on the pointer if you klick at the motortype and you can´t get rid of it.
So there sould be a Warning not to translate the string or the possibility to translate.
[image]

THEb0nny, 2020-03-29:
This method does not work for me. I cleaned the cache in Chrome.

THEb0nny, 2020-03-29:
Chrome shows not translated words, but an edge with chromium translated ...

THEb0nny, 2020-03-29:
@pelikhan need your decision...

closed 2023-05-19 by THEb0nny