microsoft/pxt-ev3 #998: Bluetooth does not work

THEb0nny, 2021-03-21:
I tried to load the bluetooth code into EV3. But now it cannot be done. An error is always thrown. Try it yourself. Something might have been updated in Chrome.
@pelikhan

pelikhan, 2021-09-10:
I'll take a peek next week.

THEb0nny, 2021-09-19:
@pelikhan hi, have you looked?

THEb0nny, 2021-09-22:
@pelikhan https://github.com/microsoft/pxt-ev3/issues/902#issuecomment-674217648

THEb0nny, 2021-09-30:
@pelikhan it turned out that with the bluetooth connection?

THEb0nny, 2021-10-10:
@pelikhan have you seen why bluetooth does not work?

dmitryaleshin, 2023-08-07:
The project is death but problem exists. I've attempted to create a PR with fix but have no proper rights.

THEb0nny, 2023-08-07:
> The project is death but problem exists. I've attempted to create a PR with fix but have no proper rights.

hello, I support the project, my changes are accepted, a new version is added.

I have looked at this problem myself. The problem on all platforms is in makecode, but the developers have no plans to fix it. For example, in microbit it is also impossible to fill in the firmware in the same way.

THEb0nny, 2023-08-07:
I didn't understand that the second part

dmitryaleshin, 2023-08-07:
> I didn't understand that the second part

Thats not a big deal. https://www.lego.com/cdn/cs/set/assets/blt6879b00ae6951482/LEGO_MINDSTORMS_EV3_Communication_Developer_Kit.pdf  the protocol mentioned the length property. My fix is to read full packet instead the buffer. This is a common problem with all the serial communication and i think it's pretty clear.

THEb0nny, 2023-08-07:
Хорошо, я тоже посмотрю, потом дам отзыв.

closed 2023-08-14 by THEb0nny