microsoft/pxt-ev3 #1002: Today's problems

THEb0nny, 2021-09-22:
I want to raise the problems that I know at the moment, to be noticed.

**1. The bluetooth stopped working.** https://github.com/microsoft/pxt-ev3/issues/998
**2. There is no field for entering the project name in the beta version.**
In the stable version, this can be done in two places. On the page with editing the code (below) and through the menu (gear icon) and the "Project settings" item.
**4. The beta version has problems with connecting sensors.**
This problem is not present in the stable version. It seems to me that the changes that were merged in the stable version are not in the beta version.
**5. Beta motors may not stop after stopping the program.**
This is not available in the stable version. Everything is right there. As well as the last point, I think that the code of changes was not included in the beta version. Maybe I'm certainly wrong...
6. The run block cannot be translated, because of this the motor selection list (A + B, B + C and other options) sticks to the mouse cursor. https://github.com/microsoft/pxt-ev3/issues/981
7. The storages extension does not work in beta. The program can no longer compile. https://github.com/microsoft/pxt-ev3/issues/983
8. Problems with the i2c protocol. i2c sensor ht color sensor v2 starts working if only to reconnect the port. https://github.com/microsoft/pxt-ev3/issues/1000

@pelikhan I would also very much like you to try to raise the ptx version. The newer versions have some nice features. For example, it is very important to see the return block for visual programming in blocks.

THEb0nny, 2021-09-23:
@pelikhan worked with kids today, motors really might not stop in beta.

pelikhan, 2021-09-23:
Beta is not ready for production. As you observed, there are issues with motors.

THEb0nny, 2021-09-24:
@pelikhan I understand this, but the beta version contains new changes. I just don't understand why I managed to defeat the stop of the motors in the stable version (as well as the flashing of the sensors), but in the beta version it does not work.

THEb0nny, 2021-10-10:
Now I translated the Run block, it works. Not buggy. Buccilic due to the fact that the transfer is added big, small? They can not translate them now

THEb0nny, 2021-10-10:
@pelikhan And why in the beta version the field below is not displayed, which is for entering the name of the project? Is this a pxt or pxt-ev3 kernel issue?

THEb0nny, 2023-05-19:
Some of the problems have been solved by me.

closed 2023-05-19 by THEb0nny