microsoft/pxt-ev3 #1010: Tilt Angle Gyro Sensor

ManagedStudios, 2023-01-22:
The newer Gyro Sensors with a date code N4 or higher have a 2 axis gyroscope inside the sensor. It is really frustrating that you can use just one of them. For detecting inclined planes for example it would be great to be able to retrieve the tilt angle from the gyro sensor. 

EV3DEV has a built in solution and EV3-G software has a third party block integration (https://github.com/dlech/EV3-G-Blocks/releases). 

Would be so great if we get this feature also in makecode

THEb0nny, 2024-01-24:
I was wrong when I wrote this. The gyroscope does not work in port 1 in the beta version, because... there is a bug with 1 port. Sensors with UART do not work in it.

THEb0nny, 2024-02-15:
@ManagedStudios https://github.com/microsoft/pxt-ev3/pull/1050