microsoft/pxt-ev3 #1026: Somtimes, I can't open the xx.py file in ev3

danke1, 2023-07-11:
In the ev3 device,
when i click the move.py file, as soon as I enter to next, I return to the list page again.
Repeat this process several times in a row

How can i handle with this?

THEb0nny, 2023-07-17:
What are py file extensions? python? How did you upload it there?

Darrenpig, 2023-11-01:
> What are py file extensions? python? How did you upload it there?

A Project called ev3dev "Hello World for ev3dev + Visual Studio Code + Python" can make it.

Darrenpig, 2023-11-01:
Starting: brickrun --directory="/home/robot/EV3test" "/home/robot/EV3test/main.py"
----------
Starting program failed: Failed to execute child process "/home/robot/EV3test/main.py" (Permission denied)
----------
Exited with error code 1.
Sometimes, I got this^