microsoft/pxt-ev3 #1032: Unable to build and deploy

dmitryaleshin, 2023-08-07:
I'm having issues running build code from readme
_cd libs/core
pxt deploy_
produces the following error for me:
_Using target ev3 with build engine dockermake
  target: v1.4.35 Y:\GitHub\pxt-ev3
  pxt-core: v7.2.27 Y:\GitHub\pxt-ev3\node_modules\pxt-core
building Y:\GitHub\pxt-ev3\libs\core
INTERNAL ERROR: Error: cannot find any drives to deploy to
    at Y:\GitHub\pxt-ev3\node_modules\pxt-core\built\buildengine.js:551:23
    at processTicksAndRejections (node:internal/process/task_queues:96:5)
PS Y:\GitHub\pxt-ev3\libs\core>_

THEb0nny, 2024-02-04:
Делайте просто
cd ../pxt-ev3
pxt serve

dmitryaleshin, 2024-02-04:
Да да. Любую проблему можно просто игнорировать. Для этого и создаются баги.

THEb0nny, 2024-02-04:
Ну эту проблему разрабы майкрософта решать не будут точно